FreeCAD/FreeCAD-macros #6: Decide which honeycomb macro to add in to repo

luzpaz, 2017-05-22:
Thread: https://forum.freecadweb.org/viewtopic.php?t=11008

galou, 2017-06-09:
I see two macros in this forum topic, one from ian.rees and one from chrisw. The one from ian.rees doesn't work on my setup. I would be glad if someone could test both and tell me whether the ian.rees' macro is better than chrisw and I would attempt to make it compatible with my setup.

OS: Ubuntu 14.04.5 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.11243 (Git)
Build type: None
Branch: master
Hash: 6e0f060e96ee54eef1355f221abf0156de45b2da
Python version: 2.7.6
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 7.1.0

galou, 2018-06-06:
There is a new player in town: https://github.com/mwganson/FCHoneycombMaker. Cf. also https://forum.freecadweb.org/viewtopic.php?f=22&t=29073. It's not Part Design compatible as far as I see from the screenshots.