FreeCAD/FreeCAD-macros #8: Standardize FC macro to have uniform structure and well commented

luzpaz, 2018-01-05:
Reasons:  
* we can run unit tests on them to see if they work  
* they are readable and users can study them  
* inspect/test them for vulnerabilities  
* remove typos  
* have the ability to eventually run them in console mode only (depending on the viability of this)  
* easy to understand attribution to the author(s)  
* clear posting of what the licence is

galou, 2018-01-05:
Macros already are supposed to have a uniform structure, cf. for example https://github.com/FreeCAD/FreeCAD-macros/blob/9916141e75c5e7463305b0481f0a5939a5cdafbb/ObjectCreation/Macro_FCCircularText.FCMacro#L56. I agree with you that the license information is missing.

Can you please provide an example of how a uniform structure could be used in unit-tests? It sounds interesting.

I don't understand however how a uniform structure could help with typos.

luzpaz, 2018-01-05:
@galou not all of them have all this info. I'll format my idea in regards what else I had in mind

luzpaz, 2018-01-23:
related: #9

luzpaz, 2019-01-17:
Add info from https://forum.freecadweb.org/viewtopic.php?f=22&t=30636