FreeCAD/FreeCAD-macros #32: Conversion/DeepCopy.FCMacro ImportError

luzpaz, 2019-01-25:
Same Error Message in Py2Qt4 and Py3Qt5, author Galou
```
Traceback (most recent call last):
  File "E:/Data/FreeCAD/Macro/DeepCopy.FCMacro", line 14, in <module>
    from freecad import app
<class 'ImportError'>: cannot import name 'app'
```

Related to #8  
Found by syres on the [forum](https://forum.freecadweb.org/viewtopic.php?f=22&t=30636&hilit=macros#p280485)

looooo, 2019-01-25:
we had discussion on this topic (lower case import-names) But it never was decided to go this way. So it's better to stay with "from FreeCAD import App".

Although this syntax works for me with:
```
OS: Ubuntu 18.04.1 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15671 (Git)
Build type: Release
Branch: master
Hash: 28cbe2bd8986701bb7404842cf60138957e23ed6
Python version: 3.6.7
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: German/Germany (de_DE)
```

galou, 2019-01-25:
I opened a topic on this, https://forum.freecadweb.org/viewtopic.php?f=10&t=33545, but it didn't attract any attention from developers.
Which ever solution we choose, there should be only one way to import FreeCAD and it should work everywhere.
Fortunately, this kind of changes is rather easy to do almost automatically. I'm waiting a bit for some reaction (or PRs with `from FreeCAD import App as app`) and if nothing comes, I'l do the changes so, that macros work everywhere, as soon as I find time and mood for them.

`from freecad import app` works also on my system:
```
OS: Ubuntu 18.04.1 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15694 (Git)
Build type: Release
Branch: master
Hash: 96dee91d5e7d7fd37f43a4dcf406a120ae1f1b8a
Python version: 2.7.15rc1
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

galou, 2019-02-01:
cf. 4c028d2.

closed 2019-02-01 by galou