FreeCAD/FreeCAD-macros #34: Addon Manager utf8 issues

luzpaz, 2019-01-25:
- [ ] Utility/HighlightDifference.FCMacro

Failed to install using Py3Qt5 AddOnManager with error :
```
Traceback (most recent call last):
  File "E:\Data\My Downloads\FreeCAD_0.18.15643_Conda_Py3QT5-WinVS2015_x64\FreeCAD_0.18.15643_Conda_Py3QT5-WinVS2015_x64\Mod\AddonManager\AddonManager.py", line 423, in install
    if install_macro(macro, self.macro_repo_dir):
  File "E:\Data\My Downloads\FreeCAD_0.18.15643_Conda_Py3QT5-WinVS2015_x64\FreeCAD_0.18.15643_Conda_Py3QT5-WinVS2015_x64\Mod\AddonManager\AddonManager.py", line 135, in install_macro
    macrofile.write(macro.code)
  File "E:\Data\My Downloads\FreeCAD_0.18.15643_Conda_Py3QT5-WinVS2015_x64\FreeCAD_0.18.15643_Conda_Py3QT5-WinVS2015_x64\bin\lib\encodings\cp1252.py", line 19, in encode
    return codecs.charmap_encode(input,self.errors,encoding_table)[0]
UnicodeEncodeError: 'charmap' codec can't encode character '\u2212' in position 2434: character maps to <undefined>
```
---
Related to #8  
Found by @Syres916 on the [forum](https://forum.freecadweb.org/viewtopic.php?f=22&t=30636&hilit=macros#p280485)v

galou, 2019-02-01:
For information, `Camera.FCMacro` installable on
```
OS: Ubuntu 18.04.1 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15713 (Git)
Build type: Release
Branch: master
Hash: 93dd00cc3953416bf46fb0a5b7b3c6397df6e4d0
Python version: 2.7.15rc1
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

galou, 2019-02-20:
This is probably related to macros from the Wiki and a problem with the AddonManager, not macros from this repository.

Feel free to reopen if required.

closed 2019-02-20 by galou