FreeCAD/FreeCAD-macros #35: Tracking: Macros not shown in the AddonManager

luzpaz, 2019-01-25:
- [ ] ObjectCreation/ParabolaCreater.FCMacro
- [ ] FCRotateViewAbsolute.FCMacro
- [ ] Foto.FCMacro (and it errors out, see below)	
```
Error:
Traceback (most recent call last):
  File "E:/Data/FreeCAD/Macro/Foto.FCMacro", line 56, in <module>
    sel0 = sel[0]
<class 'IndexError'>: list index out of range
```

galou, 2019-02-01:
For information, `ParabolaCreater.FCMacro` is installable on
```
OS: Ubuntu 18.04.1 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15713 (Git)
Build type: Release
Branch: master
Hash: 93dd00cc3953416bf46fb0a5b7b3c6397df6e4d0
Python version: 2.7.15rc1
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

luzpaz, 2019-02-02:
@galou do you see it in the addon manager menu ?

galou, 2019-02-02:
Yes, I see `ParabolaCreater.FCMacro` in the AddonManager and can install it.

Do you have `python-git` installed?

galou, 2019-02-03:
The AddonManager uses two sources for macros, this repository and the Wiki pages.
Getting macros from this "official" repository requires `gitpython` (or `python-git`, however it's called).

luzpaz, 2019-02-03:
Very good to know. We should let people know this in the Add-on Manager interface or perhaps when FreeCAD first loads? I'm using the appimage so maybe we need to bundle the appimage with `python-gitpython` ?

galou, 2019-02-04:
This is new to 0.18.

closed 2019-02-20 by galou