FreeCAD/FreeCAD-macros #37: Add Macro_DXF_to_Sketch_layers.FCMacro

galou, 2019-02-14:
https://forum.freecadweb.org/viewtopic.php?p=286770#p286770

galou, 2019-02-20:
Done in dd71b5c.

closed 2019-02-20 by galou

sliptonic, 2019-05-06:
I tried executing the macro.  I get the prompt to install using pip.  When I confirm, I get an error.  I'm on debian buster so it would require pip3 install ezdxf.

I manually installed ezdxf and tried importing a file.  I got this error:

  File "/home/brad/Dropbox/FREECAD/DxfToSketchLayers.FCMacro", line 210, in main
    unitfac = unitfacmap[dfile.header['$INSUNITS']]  
<class 'KeyError'>: (1,)

galou, 2019-05-07:
The macro is not Python3 compatible yet. Please rather open a new issue.

About the installation issue. The installation used to be done via `import pip` but @goatchurchprime changed it back to using subprocess, I guess for a good reason.