FreeCAD/FreeCAD-macros #67: Discussion: the utility of the wiki macros

luzpaz, 2019-09-15:
I'll fill this issue in with more details soon. I'm just starting it here with some placeholder text for now. 

* Find thread with mario52 and fixing typos/grammar
* Issues: the macros are in gists, why? It seems there was an issue that was encountered which influenced moving to this 'gist' fix.  
* problems with gists: 
  * not easily browesable 
  * jumping through a lot of hoops to get to (go to wiki page, expand macro, copy gist url
  * gists have multiple versions (though the latest is shown, but could be confusing)
  * gist URL is long
  * how do users know who mario52 is if they look at the http logs (better for it to be under the FreeCAD org umbrella
  * difficult to submit fixes and refinements to
  * difficult to organize

galou, 2019-09-17:
I didn't check what you write but I guess you're right about limitations of gist. What do you think about opening a new repository for macros of the wiki?
As I wrote in https://github.com/FreeCAD/FreeCAD-macros/issues/49#issuecomment-501576412, I'd like to keep this repository as clean as possible (though I have to admit that I'm not completely satisfied about the current state and I'm thinking of deleting macros with low quality or little use).

The second repository (for example named `wiki-macros`) would be just a place where to put macros for storage. Maintainers would just accept pull request without review. The only thing I would do (I count myself as a potential maintainer) would be to create directories corresponding to the categories of the macros in the wiki.

luzpaz, 2019-09-17:
hmm...I'm thinking we need more feedback from the community on this. As FC grows we do need to have a way to vet macros we offer people. For example, Arch Linux has Arch Packages and then has the Arch User Packages or AUR. There are a lot more AUR packages then there are Arch Packages, and they aren't strictly vetted with a forewarning about that. So it sounds like that is what you're proposing?

galou, 2021-01-19:
Closed because of lack of activity.

closed 2021-01-19 by galou