FreeCAD/FreeCAD-macros #68: Cannot retrieve the list of addons

KoltesDigital, 2019-12-08:
Hi,

In the addon manager, when I click on the Macros tab, it keeps showing "downloading list of macros". The report view shows:

```
Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.18\Mod\AddonManager\AddonManager.py", line 701, in run
    self.retrieve_macros_from_wiki()
  File "C:\Program Files\FreeCAD 0.18\Mod\AddonManager\AddonManager.py", line 741, in retrieve_macros_from_wiki
    u = urlopen("https://www.freecadweb.org/wiki/Macros_recipes")
  File "C:\Program Files\FreeCAD 0.18\Mod\AddonManager\addonmanager_utilities.py", line 36, in urlopen
    u = urllib2.urlopen(url, context=ssl_ctx)
  File "C:\Program Files\FreeCAD 0.18\bin\lib\urllib\request.py", line 223, in urlopen
    return opener.open(url, data, timeout)
  File "C:\Program Files\FreeCAD 0.18\bin\lib\urllib\request.py", line 532, in open
    response = meth(req, response)
  File "C:\Program Files\FreeCAD 0.18\bin\lib\urllib\request.py", line 642, in http_response
    'http', request, response, code, msg, hdrs)
  File "C:\Program Files\FreeCAD 0.18\bin\lib\urllib\request.py", line 570, in error
    return self._call_chain(*args)
  File "C:\Program Files\FreeCAD 0.18\bin\lib\urllib\request.py", line 504, in _call_chain
    result = func(*args)
  File "C:\Program Files\FreeCAD 0.18\bin\lib\urllib\request.py", line 650, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 404: Not Found
```

It seems you recently changed something to the wiki URLs, because most links found when googling are also invalid wiki URLs.

As a nice to have feature, this exception should be caught, in order to inform the user about server issues.

luzpaz, 2019-12-08:
@vocx-fc any idea?

vocx-fc, 2019-12-09:
Well, the server was down. Kremitzki fixed it already so I suppose it should work now, [freecad.org site unavailable](https://forum.freecadweb.org/viewtopic.php?f=8&p=352684#p352684).

As for dealing with the exception; well, sure I guess that could be handled better in the `addonmanager_utilities.py` code, but it doesn't seem very critical. The Python backtrace does show the HTTP Error 404, so that should be an indication that something is down.

KoltesDigital, 2019-12-09:
Great for the site availability! I confirm it's working.

It's not critical at all. However I discovered FreeCAD recently in a workshop where people were not into programming at all, which I guess is most of the target audience, so it could be nice for them.

luzpaz, 2019-12-09:
I guess we could add a more user friendly message. I'll open a ticket on FreeCAD/FreeCAD-addons

luzpaz, 2019-12-09:
@KoltesDigital please close issue. thanks.

KoltesDigital, 2019-12-09:
Thanks for your support!

closed 2019-12-09 by KoltesDigital