FreeCAD/FreeCAD-macros #101: SketcherOffset.FCMacro fals in 0.20

macdroid53, 2021-04-15:
When I execute this macro the following is shown in report view:

```
12:31:24  Traceback (most recent call last):
  File "/home/mac/SharedData/FC_common/SketcherOffset.FCMacro", line 440, in <module>
    ex = OffsetDialog() 
  File "/home/mac/SharedData/FC_common/SketcherOffset.FCMacro", line 35, in __init__
    self.tracker = DraftTrackers.lineTracker(scolor=(1,1,1),swidth=3,dotted=True) 
<class 'NameError'>: name 'DraftTrackers' is not defined

```
OS: Ubuntu 20.04.2 LTS (XFCE/ubuntustudio)
Word size of FreeCAD: 64-bit
Version: 0.20.24711 (Git)
Build type: Release
Branch: master
Hash: 47a41ae1c8a08fe47ce0e7b5b73df8f849adfe21
Python version: 3.8.5
Qt version: 5.12.8
Coin version: 4.0.0
OCC version: 7.5.1
Locale: English/United States (en_US)

galou, 2021-04-30:
PR #102

macdroid53, 2021-04-30:
Same error in today's build:

`08:03:10  Traceback (most recent call last):
  File "/home/mac/SharedData/FC_common/SketcherOffset.FCMacro", line 440, in <module>
    ex = OffsetDialog() 
  File "/home/mac/SharedData/FC_common/SketcherOffset.FCMacro", line 35, in __init__
    self.tracker = DraftTrackers.lineTracker(scolor=(1,1,1),swidth=3,dotted=True) 
<class 'NameError'>: name 'DraftTrackers' is not defined`

OS: Ubuntu 20.04.2 LTS (XFCE/ubuntustudio)
Word size of FreeCAD: 64-bit
Version: 0.20.24824 (Git)
Build type: Release
Branch: master
Hash: e2e380655bbe10af8ed6573be9b58ebc395fb841
Python version: 3.8.5
Qt version: 5.12.8
Coin version: 4.0.0
OCC version: 7.5.1
Locale: English/United States (en_US)

galou, 2021-05-02:
@macdroid53 You didn't test my changes but something else because line 35 on my version is different.

galou, 2022-02-04:
Any update on this?

galou, 2022-11-10:
Closing for lack of reaction.

closed 2022-11-10 by galou