FreeCAD/FreeCAD-macros #109: Please add circle support

Goodfeat, 2021-09-21:
Hello, 
Thank you for creating this wonderful script.

there is a request, could you please add circle support.

I tried to modify the script of the previous version.
the your latest version is quite unstable and the results are unpredictable

I used the previous version where you could manually add values, and also tried to implement additional options for disabling binding and reverse. the creation of two directions at the same time was also added.
the script is still incomplete and has some bugs

However, the main disadvantage is the inability to create an offset from the circle. 

Could you help add circle offsets or at least give me a hint
[OffsetInSketcherPlus.zip](https://github.com/FreeCAD/FreeCAD-macros/files/7199913/OffsetInSketcherPlus.zip)

galou, 2021-09-21:
@edi271 can you please have a look at this? Thanks!

edi271, 2021-09-21:
Hello Goodfeat,
please open a thread for discussing the issue in the FreeCAD forum. I do not understand, what you mean with "an offset from the cicle"

Goodfeat, 2021-09-22:
[image]
I mean, a macro cannot create offsets for circles

[image]

Apparently the script is trying to find the starting point, but it is not in the circle

p.s I will duplicate this topic in the next branch

galou, 2022-02-04:
Any update on this?

Otherwise I'll close the ticket.

edi271, 2022-02-15:
Please close the ticket

closed 2022-02-15 by galou