FreeCAD/FreeCAD-macros #113: LasercutterSVGExport does not support arrays

emutier, 2021-12-10:
Hi, 
I am currently using LasercutterSVGExport in a project which uses the Array feature from the Draft WB. Unfortunately, this is not supported by this macro. 
The macro works with the underlying part so I think using the part and the Object->Count information could do the trick. My freecad-scripting and python knowledge is rather basic/non-existent so I hope could by fixed in the future.

luzpaz, 2021-12-10:
CC @chbergmann

chbergmann, 2021-12-11:
There is a workaround: Select your Array, then go to Part workbench, Menu Part -> Compound -> Explode compound
This will create a folder with all elements inside your Array. Select all elements and then start LasercutterSVGExport

emutier, 2021-12-12:
Hi luzpaz, hi chbergman, 
thanks for your fast answers! My workaround was to clone the part. Although it is okay for the export it is sub-optimal for my model. 
As far as I have understood your code "for sel in selection" is the main loop. So before creating the offset one could check if sel is of type array. If so, assign Object->Base as sel and use Object->Count to initialize an inner for loop (else set the inner loop value to 1). The remaining code could stay the same. 

I tried it myself but failed to find out how to check for Array-Type and also failed to use the debugger. It simply ignored my breakpoints... really frustrating. I am willing to contribute but I need some push in right direction :)

off-topic: what are the chances to combine this macro with [deepnest](https://deepnest.io/) https://github.com/Jack000/Deepnest?  I think the core is written in Python and C++.

chbergmann, 2021-12-15:
I made a change: The macro checks if there are multiple solids in the selected object and creates an outline for each. This is more generic approach and should work not only for arrays. 
The macro is also present as a tool in the Curved Shapes workbench.

Deepnest cannot be integrated easily. It has to be installed or compiled on your PC first.

luzpaz, 2021-12-15:
What about https://wiki.freecadweb.org/Arch_Nest ?

emutier, 2022-06-14:
hi guys, 
sorry that I dont answer. I think Dec 2021 was the last time I opened FreeCad :(
@chbergmann thx for the changes. works perfectly fine
@luzpaz I was not able to get arch nest running. could not pick a container. wiki is not really helpful either

chbergmann, 2023-12-02:
@luzpaz The LasercutterSVGExport macro cannot be installed by the addon manager in FreeCAD. why ?