FreeCAD/FreeCAD-macros #140: Camera.FCMacro - invalid line split at 1335-6

kenhia, 2022-08-20:
https://github.com/FreeCAD/FreeCAD-macros/blob/29c73ca90f4c3716f120a2e89ebdd698bbf9a463/PureGui/Camera.FCMacro#L1335

NB: I'm a FreeCAD newbie, so I may be holding things wrong, but...

Lines 1335-6 have a bad string split:

```python
            App.Console.PrintMessage("Refresh (3Dview = X " + str(ori_Horizont) + ",
                                                        Y " + str(ori_Vertical) + ") " + "\n")
```
the final double-quote on 1335 is not closed until the next line. Fixed in my copy by just making 1335-6 into one line:

```python
            App.Console.PrintMessage("Refresh (3Dview = X " + str(ori_Horizont) + ", Y " + str(ori_Vertical) + ") " + "\n")
```
Could also be fixed by moving the `",` to 1336 to produce:
```python
            App.Console.PrintMessage("Refresh (3Dview = X " + str(ori_Horizont) + 
                                                        ", Y " + str(ori_Vertical) + ") " + "\n")
```

luzpaz, 2022-08-20:
Thanks, can you make a PR ?

kenhia, 2022-08-20:
> Thanks, can you make a PR ?

Created - https://github.com/FreeCAD/FreeCAD-macros/pull/141

galou, 2022-08-22:
Thanks for the fix!

luzpaz, 2022-08-22:
TY @kenhia and @galou (commit: f903939)  
Closed.

closed 2022-08-22 by luzpaz