FreeCAD/FreeCAD-macros #142: [3D Generator] Something else?

18turbo, 2022-09-23:
Hello,

When does the new added macro appear in the Addon Manager?.

Do I have to do something else?

Regards

luzpaz, 2022-09-26:
You have to update the addon manager cache, it should appear after that

luzpaz, 2022-09-26:
Closing. Please re-open if an issue presents

closed 2022-09-26 by luzpaz

18turbo, 2022-09-26:
Hello @luzpaz ,

I update the Addon Manager cache and it doesn't work. Since the Merged was done I'm trying and I don't see it.

That's why I asked

Regards

luzpaz, 2022-09-26:
I can't check ATM. @galou can u confirm?

chennes, 2022-09-26:
It's showing fine for me -- make sure you're searching for it by its filename ("G3D").p

18turbo, 2022-09-27:
Everyone I ask tells me they don't see it

18turbo, 2022-09-28:
it works fine. In Linux, you must have installed Gitpyhton ;)

Thanks for all!