FreeCAD/FreeCAD-macros #151: BoxCreator.FCMacro - __Date__ content has missing quote termintor character

neildarlow, 2023-01-05:
As per Issue title. The problem is obvious from looking at the __Date__ line. Addon manager throws a syntax error when the local cache is updated.

luzpaz, 2023-01-05:
Please link to the the line in question.
Please always add you full [About](https://wiki.freecadweb.org/About) info when reporting issues

neildarlow, 2023-01-05:
Not that it will help but here you go:

[code]
OS: KDE Flatpak runtime (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.6, Qt 5.15.7, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Assembly4 0.12.5
  * Curves 0.6.5
  * fasteners 0.4.52
  * lattice2 1.0.0
  * Help 1.0.3
  * ProDarkThemePreferencePack 1.0.0
[/code]

```
BoxCreator.FCMacro - line 6
-__Date__ = '2023-01-03
+__Date__ = '2023-01-03'

Report View output from Refresh local cache:
15:44:06  Syntax error while reading __date__ from macro BoxCreator
15:44:06  __Date__ = '2023-01-03
```

luzpaz, 2023-01-05:
https://github.com/FreeCAD/FreeCAD-macros/blob/9d1aced568223cf753c311085200196150297225/ObjectCreation/BoxCreator.FCMacro#L7

Making PR to fix now

luzpaz, 2023-01-05:
Thank you @neildarlow

PR #152 (2023-01-05): BoxCreator.FCMacro: fix broken syntax
Closes #151
commit 79aab8f8 (2023-01-05): BoxCreator.FCMacro: fix broken syntax

Closes #151
closed 2023-01-05 by luzpaz