FreeCAD/FreeCAD-macros #153: Freecad launched with errors; addon manager doesn't work

Ericounet, 2023-01-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Debian GNU/Linux bookworm/sid (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: French/France (fr_FR)
[/code]
```

### Subproject(s) affected?

Addon Manager

### Problem description

1) When I launch Freecad, I got this error in the terminal: 

/home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/libdrm/amdgpu.ids: Aucun fichier ou dossier de ce type

the French part means: no file or directory of this type

2) the addon manager blocks when I try to add any addon.
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-01-30:
The macro seems to be trying to use Python's text-wrapping character in its list of files, which is not allowed. That list must be a simple, single-line string, no matter how long. The parser is sort of braindead, it doesn't really understand Python, it's just looking for double quotes.

galou, 2023-02-02:
The `__Files__` metadata is correctly defined in https://github.com/FreeCAD/FreeCAD-macros/blob/master/ObjectCreation/AeroFoil.FCMacro, v. 2.0.3. Which file are you talking about?

I'll close unless @Ericounet confirms an issue.

luzpaz, 2023-02-05:
Lets close and if @Ericounet responds with a confirmed issue, they can re-open it.

closed 2023-02-05 by luzpaz