FreeCAD/FreeCAD-macros #167: Invalid license for SketcherOffset.FCMacro

chennes, 2024-02-10:
This macro contains a "__license__" entry, but does not specify a valid license there.
https://github.com/FreeCAD/FreeCAD-macros/blob/25fcaf70eb49c7ba68d1b7958404ae104e15430c/Sketcher/SketcherOffset.FCMacro#L14

galou, 2024-02-27:
@edi271 can you please react to this?

Maybe we could remove the macro since offset is now implemented natively in Sketcher?

edi271, 2024-03-01:
The Offset tool implemented in the Sketcher is working in an other way, that means, it returns in some situations other results than the macro. If you think best, please remove the macro.

galou, 2024-03-01:
I don't have anything against letting the macro here if it behaves differently. Moreover, it's a well-written macro that can serve as a reference for many things.

What about the license? I can make the change myself if you name one.