FreeCAD/FreeCAD-macros #177: Create tests for each macro in order to know if it is still functioning correctly

luzpaz, 2024-10-24:
Maybe we need to consider having tests for each and every macro that can be run to test the integrity of the macro code between version updates?

@galou thoughts ?

galou, 2024-11-25:
A good idea, for sure but developing the framework for this is out of my capacities right now.

luzpaz, 2025-05-07:
@chennes maybe something for the FPA ?

oursland, 2025-05-07:
One approach is to publish a list of public APIs and provide versioning for them.  A package can declare the API versions it is requesting and when they are no longer supported they may be identified as deprecated or incompatible without the need for executing tests.

As the APIs are already defined through the [custom binding generator](https://github.com/FreeCAD/FreeCAD/tree/main/src/Tools/bindings), it may be a relatively straightforward extension to the existing descriptions.

This however does not address the functional regressions that are not associated with API changes, for which tests may be useful, but does provide a mechanism by which FreeCAD can modernize and update the APIs with breaking changes in a controlled fashion.