FreeCAD/FreeCAD-macros #196: Macro ceased to work in Freecad Weekly Build 2025.04.21 (pre release)

gdvranjes, 2025-04-28:
When you invoke the macro and click over the eye it freezes. It doesn't occur in Freecad 1.0.0

chennes, 2025-04-29:
Which macro is this in reference to?

gdvranjes, 2025-04-29:
Please excuse me. I see now this and now understand your question. I have send you a mail before seen this but I believe your question is answered. thanks for your time and patience...

galou, 2025-04-29:
@gdvranjes Does it mean we can close the issue?

gdvranjes, 2025-04-29:
No, the problem still exist. Please see attached screenshot:

[image]

When I choose to close the macro, it does without problem and does not alter the status of the drawing tree previos to the invocation of the macro...

gdvranjes, 2025-04-29:
Well, curious thing: after using the addon manager once a week since three years ago, and reporting the addon manager being all "addons" up to date, results that my version installed (0.8.0); effectively is not the last. I have downloaded (from the wiki page) e installed manually 0.8.3 in place of 0.8.0 and the behavior is the same in both versions of Freecad: in Feecad 1.0.0 no problem, in Freecad Weekly Build 2025.04.21 the macro freezes when you click over the eye. Below is the description of my system

OS: Slackware 15.0 x86_64 (KDE//usr/share/xsessions/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41354 (Git) AppImage
Build date: 2025/04/21 18:01:03
Build type: Release
Branch: main
Hash: fd390acd53dae91898145a22d74050d0d648a45b
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: Spanish/Argentina (es_AR)
Stylesheet/Theme/QtStyle: unset/Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/52.445/1
Installed mods: 
  * Plot.backup1736919899.3132463 2024.11.26 (Disabled)
  * CurvedShapes.backup1736919830.8936977 1.0.13 (Disabled)
  * AirPlaneDesign 0.4.1
  * Curves.backup1736919830.987689 0.6.53 (Disabled)
  * OpenDark.backup1725671265.1599011 2024.3.13 (Disabled)
  * CurvedShapes 1.0.13
  * freecad.gears 1.0.0

gdvranjes, 2025-04-29:
And there is more: I have invoked the addon manager again and it had replaced version 0.8.3 of the macro manually installed with 0.8.0 again... So as being things, every time I invoke the addon manager, it will return to the 0.8.0 version of the macro overwriting the manually installed version...

gdvranjes, 2025-04-30:
Same behavior with Weekly Build 2025.04.28 (pre release)...

gdvranjes, 2025-05-07:
Good night. Update about the macro problem with development versions of Freecad:
If you invoke the macro within a drawing containing a folder in the tree structure named CenterOfMass (result of the use on a previous version of Freecad...) **the macro work as it should!**.
But try now to click on the view button (disable the viewing of the macro results on the drawing...). The folder CenterOfMass dissapears from the tree structure, all products of the macro are gone on the drawing, **and again the freeze occurs**.
On Freecad 1.0.0 you can resume the macro use without problems.

gdvranjes, 2025-05-07:
What returns to the first manifestation of the problem: you have a new drawing, and use the macro within a development version of Freecad for the first time (No previous CenterOfMass folder created for obvious reasons...), click on the view box and now you have a freezing.