FreeCAD/FreeCAD-macros #201: Macros without licenses

chennes, 2025-08-24:
There are currently six macros in the repository here that don't have a license set in `__LICENSE__`: without it, they are "All rights reserved" and we don't actually have the right to distribute them. This is almost certainly not the author's intent: we should reach out to the authors of the following macros and ask them for licensing information. Without it, we should remove them from the repo.
* Foto
* FCRotateViewAbsolute
* SketcherClipView
* SketcherOffset
* Camera
* AirfoilImportAndScale

oursland, 2025-08-26:
@quick61 Could you update the license for `AirfoilImportAndScale`?

chennes, 2026-01-06:
@galou AirfoilImportAndScale is listed on the Wiki, so we can use CC-BY-3.0 as its license. Foto and FCRotateViewAbsolute don't, and so should probably also be removed.